sb53systems/G-Code-Flow-Temperature-Controller #1: Is temperature the best variable to monitor? Or is energy?

Ergonomicmike, 2024-08-17:
This is not an Issue. It's for "Discussion." Just brainstorming for a possible Version 2.0.

I think that you are making a good start here with your Temperature Controller. And it might be that nothing more is needed.

My comment is that using temperature as a metric for materials that undergo phase change (or other type of non-linear or non-monotonic change) might not be the best variable to monitor/control. (Although it is the easiest and might be good enough for what we are trying to achieve here.)

As a simple example, when water is heated to boiling, the temperature will remain at 100 degrees (C) as the liquid changes to a gas. 

So if one were to measure only temperature while boiling water - and try to control it - one might conclude that nothing is changing. Or that nothing needs to be changed when the temperature stays at 100 degrees. Whereas energy still needs to be added to the liquid to make it boil.

The point is that temperature does not tell the whole story.

I'm thinking that after your Controller is operating well using temperature, that perhaps "energy" (joules) is what should be controlled/calculated. Or maybe even "power." 

For example, faster flowing filament requires more energy to keep the extrusion at the phase/viscosity that we want it. 

It requires more energy because there is more filament being melted per unit time (power) and also because more energy is being carried away (from the extruder) by the (melted) filament at high flow rates.

So the PWM to the heater should increase just before high filament flow and continue high until just before the filament flow returns to nominal. But how much energy should be added?

Now, it might be in the above example that even if we increase the energy to the heater, the measured temperature of the filament (actually, of the heat block) might not change in the short term. Especially if the heat block has a lot of thermal inertial. 

And so I am thinking that calculated energy or power needed is a better metric for consistent flow/appearance.

Unfortunately I'm not smart enough to do the math. But I offer the idea for a Version 2.0 if desired.

Nathan22211, 2024-08-17:
I don't think 3d printers melt plastic to a liquid state though. There are some that change structure when hot enough e.g. PET, HDPE.

sb53systems, 2024-08-17:
Hey,
I'm not a specialist in polymer extrusion, but I think both solutions will do the same!  
the filament absorbs the energy gradually in the heating block until it comes out of the nozzle with a final temperature (printing PLA at 250/270° and Height speed, does not mean that it has reached this temperature), it is necessary to calibrate the temperatures used according to the appearance or behavior of the extruded filament, not according to the maximum flow.  
  
Thank you for your participation.  
Happy Sart 3D Printing.
Salim.

Ergonomicmike, 2024-08-17:
Since I am brainstorming, another possible way to feed-forward this is to use Klipper's feature that provides the Temperature State (of the extruder), expressed as a percentage of power. (PWM I presume.) 

A user would take two measurements: 1) The average PWM percentage at a set temperature (say 220 C for PLA) after steady state PWM is reached. Call this the "idle" PWM value. (PID tuning might affect how much this value perturbs around average.)

2) Then the user extrudes PLA at max flow rate, say 12 mm^3/sec for, say, 30 seconds. (Or until steady state PWM is reached.) The average PWM at the end of the test is the "max" PWM value. (A user could use a snippet from Stefan's code to perform this test.)

So then, for sections of a print that will print at the max flow rate, we would command the max PWM value for that section. (Probably with a lead-in and lead-out time to counter thermal inertia.) 

For a section with half the max flow rate, then we would command the PWM to be = idle PWM + ½ max PWM. (Here I'm assuming a linear relationship between extra energy needed (above idle) and flow rate. I don't expect that it's linear because real life doesn't scale linearly with volume. But a linear approximation might be good enough here for what we want to accomplish.) 

If one were to use PWM values, then perhaps Energy Compensation should happen in Klipper, as opposed to happening in slicers. (Because slicers can't control PWM. But Klipper can.)

sb53systems, 2024-08-17:
@Ergonomicmike, I thank you for these interesting proposals, and encourage you to check the feasibility of the PMW change during printing, direct me if so.  
I'm not as good a programmer and I can't stay much time on this project, because it's supposed to be implemented in Slicers.  
This project is just supposed to take the first step and show the benefits.  
I remain at your disposal if you need more information.  

Best regards.
Salim.

Ergonomicmike, 2024-08-18:
@sb53systems No worries. My hope is that someone from Klipper will see your work and my idea and, if possible, use them to improve 3d printing for all of us.

Nathan22211, 2024-08-23:
> <sb53systems/G-Code-Flow-Temperature-Controller/issue/1/issue_event/13937054371
> @github.com>

Found this out, there is a max power param to set heater temperature.
Danger klipper can also change PID values, though it's MPC feature might
fare better with this script.

Also any chance of Linux support. I don't think Pascal is hard to compile
on Linux so I'm tempted to do it myself

sb53systems, 2024-08-23:
Let me know if you have trouble compiling on other platforms.
Good luck.