sb53systems/G-Code-Flow-Temperature-Controller #2: Taking forever to generate G-code

TipicFaker, 2024-08-17:
When I click auto-adjust temperature or click proceed, the red text at the bottom shows generating or 0 minutes and the save and close button cannot be clicked.

In spanish, it says: " is not a valid floating point value.

TipicFaker, 2024-08-17:
I am going to check the configs for any value that's empty

TipicFaker, 2024-08-17:
This is my config.json:

```{
  "max_velocity": 200.0,
  "max_acceleration": 10000.0,
  "minimum_cruise_ratio": 0.5,
  "square_corner_velocity": 5.0,
  "instant_corner_velocity": 1.0,
  "mm_per_arc_segment": 1.0,
  "move_checkers": [
    {
      "axis_limiter": {
        "axis": [
          0.0,
          0.0,
          1.0
        ],
        "max_velocity": 5.0,
        "max_accel": 100.0
      }
    },
    {
      "extruder_limiter": {
        "max_velocity": 53.21621607382886,
        "max_accel": 2660.8108036914427
      }
    }
  ]
}```

TipicFaker, 2024-08-17:
I am using bambu studio btw

sb53systems, 2024-08-17:
Note that this script is not compatible with the bambu Start Print G-Code instructions.
You can email me your g-code file, which I can investigate further.

Best regards.
Salim.

closed 2024-08-19 by sb53systems