sb53systems/G-Code-Flow-Temperature-Controller #3: RepRap Compatibility?

basic-tv, 2024-08-19:
Any way to get this to work with a RepRap machine?

I replicated the printer config file to match my RepRap printer's settings. But when I go to create a gcode file I get Error I/O 32.
Any future plans on making a RepRap version?

sb53systems, 2024-08-19:
be aware that temporarily this is not planned, because I only have one test printer running on Klipper.  
Be aware that the working principle of this script is not as complicated as it appears, and that the only way to make it work on other machines or firmware is that this concept is integrated into the Slicer.  
if you have good knowledge of the GCode, you could tinker with the Gcode generated by the script in order to make it work on Reprap printers.  
I could help you with this if you send me an example of G-Code generated by a Slic3r Slicer.  
  
Best regards.  
Salim.

sb53systems, 2024-08-19:
it also remains to calculate the printing time according to the kinematics used in RepRap firmware !

basic-tv, 2024-08-19:
Hmm. Maybe Ill dive into it. I'd like to be able to integrate it in Orca and Simplify for RepRap machines without editing Gcode scripts.

Maybe in the near future I'll dive into that and help make a RepRap version.

sb53systems, 2024-10-01:
@basic-tv There are those who were able to make it work under RRF, by changing the parameters of the config.json file by their equivalent values.  
The time estimate is not very accurate, but the result obtained is good !  
  
Please note that you must use the latest script update.  
  
Let me know if you need more information.  
  
Best reguads,  
Salim

sb53systems, 2024-11-04:
@basic-tv the current version of the script is supposed to be usable for all firmwares, could you confirm that it is functional with your printer?
Let me know if you need more information