sb53systems/G-Code-Flow-Temperature-Controller #4: Is there any way to get it to work with START_PRINT instead of PRINT_START

skullbone, 2024-08-19:
Simply put, my creality K1 uses START_PRINT instead of PRINT_START, but the extruder and bed temp g-code are the same. I'm just wondering if there's a workaround to this because I would like to try this out

sb53systems, 2024-08-19:
you can change the name of this macro easily in the Firmware!  
Do you have any restrictions in your printer's firmware?

sb53systems, 2024-08-19:
Maybe I should include in the next version a field to give the user the possibility to choose the name of this macro!

skullbone, 2024-08-19:
Yeah, that would be nice after digging through configs I have found the macro and I have duplicated it and changed the name, but it would be nice to keep it stock.

sb53systems, 2024-08-19:
@skullbone Could you confirm if this issue can be Closed ?

closed 2024-08-20 by skullbone