sb53systems/G-Code-Flow-Temperature-Controller #5: error occurred

mickeyboss123, 2024-08-21:
error occurred
[image]

sb53systems, 2024-08-21:
Try using a period instead of a comma in the PA value  
[image]

sb53systems, 2024-08-21:
Which slicer did you use?

boromyr, 2024-08-25:
Same problem with the dot, the gcode comes from orca.

sb53systems, 2024-08-25:
can any one email me the original G-Code ?  
I think the problem may be caused by deferent versions of Windows in regional format !

sb53systems, 2024-08-25:
My Discord is : sb53systems

boromyr, 2024-08-25:
[Bridge_test.gcode.txt](https://github.com/user-attachments/files/16741675/Bridge_test.gcode.txt)
Windows 11 IT

sb53systems, 2024-08-25:
[image]  
same problem with me !

sb53systems, 2024-08-25:
i'll informs you more when i get it

sb53systems, 2024-08-25:
i think that i dont take in concidiration the Wipe option  
  
[image]  
  
@boromyr can you test it without wipe ?

boromyr, 2024-08-25:
same problem
[Bridge_test.gcode.txt](https://github.com/user-attachments/files/16741714/Bridge_test.gcode.txt)

sb53systems, 2024-08-25:
@boromyr use : PRINT_START instead of START_PRINT  
  
[image]  

You have to change it also in Klipper Configs.
Confirm if it is working.

Salim.

boromyr, 2024-08-25:
It works, I managed to save the gcode. I would like to avoid changing the name of the macros not knowing how the printer may react, so I tried:
```py
START_PRINT EXTRUDER_TEMP=220 BED_TEMP=55
PRINT_START EXTRUDER_TEMP=220 BED_TEMP=55
```
and it still works, the printer should ignore the macro with an unknown name. Wouldn't it be better to modify your software? Anyway, I will try it as soon as possible, it seems like a great improvement!

sb53systems, 2024-08-25:
your Klipper will not reconize this new macro, you have to change it

sb53systems, 2024-08-25:
@boromyr , Note that you must print each test separately to experiment with a deferent bridge speed, otherwise the script will take into consideration the average of all your examples and adjust the speed for each of them.  
  
[image]

sb53systems, 2024-08-25:
This approach will be more effective if this concept is integrated into the slicer !  
You need to experiment more with this script !
  
Happy Smart 3D Printing.