sb53systems/G-Code-Flow-Temperature-Controller #6: Can bambulab be used Flow-Temperature

yinbaozong, 2024-08-22:


sb53systems, 2024-08-22:
I didn't understand the question!  
the current version of the script is only compatible with Klipper printers and Slic3r based Slicers, like : OrcaSlicer, PrusaSlicer, SuperSlicer,,...  
  
Tell me if you want more informations.

Salim.

yinbaozong, 2024-08-24:
Thank you very much for your reply. I am using the bambulab A1 printer, and I would like to use the plugin you developed. However, based on your mention that it only supports klipper firmware, I don't think I can use it. Will you be adding support for bambulab printers in the future?

yinbaozong, 2024-08-24:
Hello, I would like to ask how to solve this “error Erreur E/S 32.”
[image]

sb53systems, 2024-10-01:
> Hello, I would like to ask how to solve this “error Erreur E/S 32.  
  
This problem is solved : https://github.com/sb53systems/G-Code-Flow-Temperature-Controller/issues/7  
  
> Will you be adding support for bambulab printers in the future?  
  
Maybe yes.  
  
Best reguards.  
Salim

sb53systems, 2024-11-04:
@yinbaozong the current version of the script is supposed to be usable for all firmwares, could you confirm that it is functional with your printer?
Let me know if you need more information

keithwkx, 2024-12-13:
Hi @sb53systems i tried using it on my bambu printer, and it seams its persistently asking for a printer config file. May i ask if there is a way for me to obtain this or an idea of what am i looking for? 
[image]

keithwkx, 2024-12-13:
can the export preset settings in orcaslicer be utilized here? 
[image]

sb53systems, 2024-12-13:
@keithwkx 
> can the export preset settings in orcaslicer be utilized here?  

is this for the config.json file?

sb53systems, 2024-12-13:
Hey @keithwkx the config.json file can be obtained automaticaly only with Klipper Firmware. For Bambu 3D printers, you need to make this file manualy.  
I can help you with that, just send me any .3mf file then i'll extract thous values from your OrcaSlicer Printer Config. example bellow:  
[image]

keithwkx, 2024-12-13:
I see, here is a .3mf file i exported.
Does this file differ for the different printer profiles i use &
Is there a guide on how do i make it manually? 
Much appreciated
[bambutest.zip](https://github.com/user-attachments/files/18125622/bambutest.zip)

sb53systems, 2024-12-13:
Here is a config.json file for your P1S  
[config.json](https://github.com/user-attachments/files/18125701/config.json)  
You have to use this button to add it  
[image]
  
Just follow this guide to make the config.json file :  
[image]

keithwkx, 2024-12-13:
TQVM, i shall try it after

keithwkx, 2024-12-13:
@sb53systems Another minor question,

[image]
the only settings i need to modify here are the filament settings right?

So for example these are my filament settings
[image]
The input should be as follows?
[image]

sb53systems, 2024-12-13:
@keithwkx You have to make a new orca profil that includes all my recommendations (Max Speed, temp-to-edit and print-end comments in the Machine G-Code).  
For the script you have to follow this guide:  
[image]  
  
I think you have to use a more gradual temp change in the script (example bellow), or make the visual calibration for your filament!  
[image]  
  
You can reach me via discord if you need more help and instructions.

zubamse, 2025-07-16:
Would be awesome for us that is not that sharp in this but still would try to use it to make an installation video/guide for bambulab's printers and recomended settings, because meny people have their printers :)

Akyariss, 2025-07-16:
With my A1 mini i'm getting great results using a 0.2mm nozzle, but i'm struggling to calibrate the 0.4mm one, especially since the PA can't be changed based on the temperature on bambulab printers (i think).

Still haven't figured out how to set a specific temperature for the first layer though