sb53systems/G-Code-Flow-Temperature-Controller #7: e/s 32 issue :

proof0zone, 2024-08-23:


sb53systems, 2024-08-23:
Send me your G-Code File at : sb53systems@gmail.com

gueee, 2024-08-24:
Got the same error when hitting  the Generate button.
Do you want me to send the unaltered g-code file, since there is no output yet, or is there?

sb53systems, 2024-08-25:
Sorry for the late reply, you can send me by email the Original G-Code, I'll let you know if I discover anything.  
  
Salim.

sb53systems, 2024-09-03:
Is this problem still present?

gueee, 2024-09-04:
> Is this problem still present?

Sorry, I couldn't find the time to retest it, but I'm going to today and will report.

sb53systems, 2024-09-04:
Know that a new version was released today :  
https://github.com/sb53systems/G-Code-Flow-Temperature-Controller/releases/tag/V1.1Beta

Nathan22211, 2024-10-01:
I just got the issue myself today

sb53systems, 2024-10-01:
@Nathan22211 This problem occurs with the integration of Klipper_Estimator with this script, you have to reboot your computer or close manualy all CMD Consol in the Task Manager.  
[image]  
  
Let me know if you manage to resolve this issue.  
  
Best regards,  
Salim

Nathan22211, 2024-10-01:
killing the console host nor rebooting fixed it.

Nathan22211, 2024-10-03:
I do get an error about a float needing to be an int in french before this occurs though

sb53systems, 2024-10-03:
> I do get an error about a float needing to be an int in french before this occurs though

I thought this error is fixed in the current version of the script!  
If you are on the latest version and this problem persists, send me the original G-Code and a screenshot of the error message.  
I'll do my best to resolve this essue.  
  
Best reguards,  
Salim

Nathan22211, 2024-10-03:
[image]

tested g code: https://gist.github.com/Nathan22211/fde16d1b3ca701afc340c66b8f25c0a2

sb53systems, 2024-10-03:
@Nathan22211 Another script modified the G-Code before this one, the Script is not yet optimized to avoid comments added in the same line.  
[image]  
  
Try one of these solutions temporarily:  
- Use this script first.
- Avoid comments in the same line in the other script.  
  
I will try to solve this problem soon in the future updates.
If you need more support, contact me in Discord: sb53systems  
  
Best reguards,  
Salim

Nathan22211, 2024-10-03:
yeah just tried it straight out of orca slicer with it being the only post script and it still errors. I am a few versions behind and on a fork so maybe I should try to use mainline

sb53systems, 2024-10-03:
> yeah just tried it straight out of orca slicer with it being the only post script and it still errors. I am a few versions behind and on a fork so maybe I should try to use mainline

@Nathan22211  Yeap, Forked version can cause essues, i recommend you use an official version.

Nathan22211, 2024-10-03:
yeah just tried it with the latest mainline beta and still the same issue when running it from orca, and I'm pretty sure I turned verbose off so there shouldn't be as much of a comment issue. also it doesn't seem to find the extruder and used filament in the g code

Nathan22211, 2024-10-03:
[image]
[Bottom Panel_GPWC_R1_ABS_1h16m.txt](https://github.com/user-attachments/files/17249587/Bottom.Panel_GPWC_R1_ABS_1h16m.txt)

sb53systems, 2024-10-03:
From where (which parameter) this comments come from !  
[image]  
I'm using Orcaslicer v2.2.0-beta2!  
[image]

Nathan22211, 2024-10-03:
verbose gcode if I recall correctly. must have left it on by accident, though some other post scripts need it turned on so there's already a conflict there.

Nathan22211, 2024-10-03:
turning verbose off fixes it but I'd definitely make it practically ignore anything after a ; in a line within the gcode blocks. some scripts, such as gradient infill, need it turned on to work properly

sb53systems, 2024-10-03:
@Nathan22211 I will try to prepare an update as soon as possible, thank you for your contribution.  
Let me know if you found otherthing to improve.