sb53systems/G-Code-Flow-Temperature-Controller #10: Error : '' is not a valid floating point value. when click on "Generating"

kristof33, 2024-09-02:
Hello,

When I click on "Generating", I have this Error : '' is not a valid floating point value. (in french : '' n'est pas une valeur en virgule flottante correcte.)

[image]

[image]

It seems to be a problem with regional parameters : . or , ?

Thanks so much for your helps.

sb53systems, 2024-09-02:
@kristof33 you have to change the name of the print start macro to : PRINT_START, you have to set the same name in klipper .cfg file too.

Best Regards.Salim.

sb53systems, 2024-09-02:
Let me know if you have trouble.

kristof33, 2024-09-03:
> @kristof33 you have to change the name of the print start macro to : PRINT_START, you have to set the same name in klipper .cfg file too.
> 
> Best Regards.Salim.

That works now.

Thanks for the clue.

closed 2024-09-03 by kristof33