sb53systems/G-Code-Flow-Temperature-Controller #11: Is it possible to change the encoding for saving gcode to UTF-8?

wtda, 2024-09-09:
In Chinese system, it seems that the generated gcode is not saved in UTF-8 format, resulting in no data underneath, and klipper can't recognise it and print it (can't see the thumbnail). Manually change ‘Output.gcode’ to UTF-8 format before it can be recognised by fluidd (can see thumbnails) and printed.
Is it possible to change the saved encoding to UTF-8?
Translated by DeepL :)

wtda, 2024-09-09:
update:
When I enable:“Beta: Use Unicode UTF-8 to get global language support”, it works!
Another question:
“UsedFilamentandExtruderintheG-Codefilearenot found ! choose your settings and click on the Refresh and generate button”
How do I include this data when slicing？

sb53systems, 2024-09-09:
> update:
> When I enable:“Beta: Use Unicode UTF-8 to get global language support”, it works!
> Another question:
> “UsedFilamentandExtruderintheG-Codefilearenot found ! choose your settings and click on the Refresh and generate button”
> How do I include this data when slicing？

Which slicer are you using ?

sb53systems, 2024-09-09:
Orca Slicer Include this information in the G-Code  
[image]  
[image]  
  
in the script, you have to create the preset with the same name as in the Slicer  
[image]

sb53systems, 2024-09-09:
send me a g-code example file with characters you use.  
I would do my best to correct the problem

wtda, 2024-09-09:
Thanks for the reply.
Comparing your screenshot with my own file, I found the problem: I have a TradRack multicolour system, so I have multiple filament name after ‘filament_settings_id’
When I use only one filament, it works fine~

sb53systems, 2024-09-09:
@wtda , If you managed to get the script to work with your conditions, when you have time, could you summarize the steps to follow ? for someone who encountered the same problem.  
  
Since this script is only an example solution, and must be developed and integrated into the Slicers, I can't adapt it to all cases. 
But if you think any improvement is required at the earliest, let me know and I'll add it to the priority list. Thank you.  
  
Best regards.  
Salim.