sb53systems/G-Code-Flow-Temperature-Controller #12: Cut window, impossible to resize.

boromyr, 2024-09-09:
I think it's a DPI issue, but the 1.1 window is too big for the screen and not scalable, the bottom part is not accessible. I'm using a 125% DPI and a 1920x1080 resolution.
[image]

sb53systems, 2024-09-09:
@boromyr yes, you need to use 100% DPI to display the interface correctly (I have the same 1920 1080 resol).  
[image]
[image]

boromyr, 2024-09-09:
Isn't very convenient, do I have to change the dpi, log out and log back in every time to use the program?

sb53systems, 2024-09-09:
Sorry about that, I'm working on this script alone. For the moment I've done the most important things, maybe later I'll add more.

sb53systems, 2024-09-09:
Also be aware that this script is not supposed to be a definitive solution, it is an Exampl of use.  

Happy 3D Printing.

sb53systems, 2024-09-09:
May add an option to hide or show the generated G-Code ?!

boromyr, 2024-09-09:
Do you mean a wrapping of the two graphs with an arrow, to expand and reduce? Like this:
[image]
It could solve the problem, I reduce a graph to generate the gcode or to look at the other graph.

sb53systems, 2024-09-12:
The Updated version of Sept 11 2024,  has the ScrollBar (Vert and Hori)  
[image]
[image]  
  
I think this will resolve your problem.  
  
Best regards.  
Salim.