sb53systems/G-Code-Flow-Temperature-Controller #14: Marlin Printer Support Please

corndog2000, 2024-10-01:
This is a really exciting advancement in 3D printing. I do wish for Marlin support so I can use it on my Prusa MK4S.

sb53systems, 2024-10-01:
@corndog2000 I am also interested in making this script Marlin compatible, but I lack a test printer that works under Marlin.  
  
I am missing the algorithm to calculate the precise time for each movement in the G-Code (for each axis: X, Y, Z, E), I use a very precise tool for Klipper Firmware +/- 1S. (Klipper_Estimator) https://github.com/Annex-Engineering/klipper_estimator  
  
And since the mechanisms of the different Firmware are almost the same, we would just have to change the equivalent variables like: Jerk / Juction deviation / Square corner velocity... (Remains to be confirmed with Marlin, other people have managed to do this with RRF).   
  
Could you send me a .3mf file generated by Orcaslicer, to allow me to analyze the Slicer parameters vs the G-Code output.  
  
And if you could help me by doing some tests on your printer, it will speed up the process.  
  
For a better collaboration, I suggest you contact me by Discord: sb53systems  
  
Best regards.  
Salim

corndog2000, 2024-10-16:
I sent you a friend request on Discord to help work on this.

sb53systems, 2024-11-04:
@corndog2000 the current version of the script is supposed to be usable for all firmwares, could you confirm that it is functional with your printer?
Let me know if you need more information.