sb53systems/G-Code-Flow-Temperature-Controller #15: temps are stuck in PLA range.

Nathan22211, 2024-10-03:
it seems the modified gcode doesn't account for nor offset for the minimum temp set in the slicer or the post-processor. I have the range set from 245-280 C and it's going down to 196C

[image]
LMK if it's an error on my part, I may be doing something wrong within orca slicer that causes this

sb53systems, 2024-10-03:
@Nathan22211 You are supposed to use the lowest temperature for 1mm3/s Flow, and the Highest Temp for you max flow.  
The temperature is estimated by the script in a linear way between (Middle Temp to Low Temp ) and (Middle Temp to Max Temp), the script will keep decreasing or increase the temperature if the flow is lowest or heigher thant the indicated flow rate in the fields.  
      
[image]  
  
I remain at your disposal if you need further information.  
  
Best reguards,  
Salim

sb53systems, 2024-10-03:
I will add a calibration section later, which explains the calibration process with 3 to 5 prints,   
I would also add a youtube video with a theoretical part and a practical part.  
  
let me know if you need further information.  
  
Best reguards,  
Salim

Nathan22211, 2024-10-03:
ok, well is the generated code meant to be in the negatives sometimes? 
[image]

sb53systems, 2024-10-03:
It's just a display bug present in the current version, look at the chart curve instead. The G-Code is correct.  
I recommend you to use 1mm3/s  as min Flow (Next version this value will be fixed).  
  
Salim.

sb53systems, 2024-10-04:
> I will add a calibration section later, which explains the calibration process with 3 to 5 prints,  
  
https://github.com/sb53systems/G-Code-Flow-Temperature-Controller?tab=readme-ov-file#ideal-flowtemperature-calibration

sb53systems, 2024-10-04:
Thanks for your feedback, I would have liked to diagnose this case, could you send me the generated G-Code.  
  
Salim.

sb53systems, 2024-10-04:
An update will be posted in a few hours, which will solve many of your issues.  
Please let me know if you have any other tips.  
  
Salim

sb53systems, 2024-10-04:
The negative temperature problem needs to be fixed, I'm trying to find its cause.  
  
here is the section of the code that is responsible for the temperature change  
[image]  
  
Please feel free to help me if you have any programming knowledge.  
  
Salim.

Nathan22211, 2024-10-04:
the only place I can think to look would be probably the getNextTemp math, that seems to be the only thing that can make a negitive. what I'd do is have a check in place to make sure nothing goes below the minimum temp, cus so far it seems there is no check there.

Nathan22211, 2024-10-04:
yeah now I'm rather stumped. I'm not too familar with pascal syntax.

Nathan22211, 2024-10-04:
it did only occur once so maybe it was a fluke? I hadn't had it occur again but I'll make sure to save the generated gcode next time it does

sb53systems, 2024-10-04:
I think I was able to fix this problem here:  
[image]  
  
I was even able to avoid comments added in the same line.  
I'm able to use your G-Code ( https://github.com/sb53systems/G-Code-Flow-Temperature-Controller/issues/7#issuecomment-2391957604 )  
[image]

sb53systems, 2024-10-04:
Could you send me another G-Code File ?

sb53systems, 2024-10-04:
Here is the new updated file : https://github.com/sb53systems/G-Code-Flow-Temperature-Controller/blob/main/bin/SB53-Systems.exe  
Note that you need to replace this file only.  
Let me posted.  
  
Salim.

sb53systems, 2024-10-04:
> I was even able to avoid comments added in the same line.  
  
[image]

Nathan22211, 2024-10-05:
seems to work better. I will note that my temps can hardly keep up with the constant changes (Rapido v2 HF with CHT extension & nozzle) on my trident even at 15K accel, which is probably low considering it's 48V on the X, Y, and extruder. No clue if using Danger Klipper's MPC is making it worse or not, but it seems this script is more optimized for traditional PID on the extruder than MPC

sb53systems, 2024-10-05:
Did you use the recommended settings in the Slicer? send me an example of a .3mf file.  
I couldn't try it on a printer other than mine, I'm waiting for your feedback and I'll try to help.  
  
How much do you rate the result on your printer?

sb53systems, 2024-10-05:
If the temperature can't keep up, you just need to reduce the heat/cool down time in the script.  
Example : from 1c°/1s to 2c°/3s

sb53systems, 2024-10-05:
Also note that this script is not specifically optimized for PID or MPC, you have to experiment yourself, and find the values ​​that suit. I think both are usable.  
  
I would be happy if people could share their experiences and settings, so that others can learn from them.  
  
I remain available to everyone if needed.  
  
Salim.