sb53systems/G-Code-Flow-Temperature-Controller #16: Error "Fichier introuvable"

Audertrak, 2024-10-11:
Windows 11
Orca Slicer 2.1.1, though I am having the issue trying to load a gcode file manually outside of the slicer

I get the following error no matter how I try to process a file
[image]

Filepath to script (OrcaSlicer does find and open the script)
C:\Users\tcunkle\Documents\3D Printing\scipts\SB53_G-Code_Flow_Temperature_Controller_V1.1Beta\SB53-Systems.exe

Machine start G-Code in OrcaSlicer
M104 S0
M140 S0 
PRINT_START BED=[first_layer_bed_temperature] 
EXTRUDER_TEMP=[first_layer_temperature[initial_extruder]]

PRINT_START macro
[gcode_macro PRINT_START]
# add PRINT_START to 'custom g-code' inside of the slicer
gcode:
    # set variables
    {% set BED_TEMP = params.BED|int %}
    {% set HOTEND_TEMP = params.EXTRUDER_TEMP|int %}
    
    # print message
    M117 SETTING START PARAMETERS
    # set bed temp
    M140 S{BED_TEMP}
    # set hotend temp
    M109 S{HOTEND_TEMP}
    
    M117 HOMING ALL AXES
    # absolute coordinates
    G90
    # home all axis
    G28

    M117 PRE HEATING
    # wait for bed to reach temp
    M190 S{BED_TEMP}
    # wait for hotend to reach temp
    M109 S{HOTEND_TEMP}

    m117 SETTING BED MESH
    BED_MESH_CALIBRATE ADAPTIVE=1

    M117 NOZZLE PURGE AND PRINT START
    # move to X10, Y10, Z0.3 at feedrate 5000
    G1 X10 Y10 Z0.3 F5000
    # zero the extruder
    G92 E0
    # extrude 15mm of filament while moving to Y160 @ feedrate 600
    G1 Y160 E15 F600
    G90

    G1 F9000
    M117 PRINTER GO BRR...

    
OrcaSlicer profile names  
[image]

Application settings
[image]

Folder contents
[image]

config.json
{
    "max_velocity": 512.0,
    "max_acceleration": 5120.0,
    "minimum_cruise_ratio": 0.5,
    "square_corner_velocity": 5.0,
    "instant_corner_velocity": 1.0,
    "mm_per_arc_segment": 0.1,
    "move_checkers": [
        {
            "axis_limiter": {
                "axis": [
                    0.0,
                    0.0,
                    1.0
                ],
                "max_velocity": 5.0,
                "max_accel": 200.0
            }
        },
        {
            "extruder_limiter": {
                "max_velocity": 64.0,
                "max_accel": 2660.8108036914427
            }
        }
    ]
}

sb53systems, 2024-10-11:
@Audertrak This is caused by the space present in the G-Code file name, or the Script path.  
[image]  
  
I remain at your disposal if you need more support.  
  
Best reguards.  
Salim.

Audertrak, 2024-10-11:
I tried removing the space from the file name with no luck. Putting the script in a path that had no spaces resolved the issue.

closed 2024-10-11 by Audertrak