sb53systems/G-Code-Flow-Temperature-Controller #17: I need moonmaker authentication.

StimeKe, 2024-11-02:
My Moonmaker is configured with authentication, and I found in the documentation that it's possible to pass an API key. Therefore, I would like to configure the API key. Although I am a developer, I am not familiar with your language, but I suspect the modification should be as follows.

Additionally, I would prefer to input a URL rather than just an IP address that you then concatenate with "http," as I might be accessing via HTTPS, especially since I might be remotely accessing my Klipper.

Code:
```
procedure TForm2.BitBtn1Click(Sender: TObject);
var
  klipperEstCom: string;
begin
  klipperEstCom:=GetEnvironmentVariable('WINDIR')+'\System32\cmd.exe /C '+ExtractFileDir(Application.ExeName)+'\klipper_estimator.exe --config_moonraker_url '+edit1.Text+' --config_moonraker_api_key ' +edit2.Text+' dump-config > '+ExtractFileDir(Application.ExeName)+'\config.json';
  ExecNewProcess(klipperEstCom,true);
  if FileExists(ExtractFileDir(Application.ExeName)+'\config.json')=true then
    if FileSize(ExtractFileDir(Application.ExeName)+'\config.json')>0 then showmessage('Printer File Saved !')
      else showmessage('Error ! File not Saved !')
  else showmessage('Error ! File not Saved !');
end;
```

sb53systems, 2024-11-02:
@StimeKe , Thank you for this remark,  
I will prepare a test version for you, (it will take me a few minutes)  
I'll wait for your confirmation that it works before sharing it with everyone.  
  
Best reguards,  
Salim

sb53systems, 2024-11-02:
Here is a modified version, I await your feedback.  
[SB53-Systems.zip](https://github.com/user-attachments/files/17607869/SB53-Systems.zip)  
  
### Edited Code
```
procedure TForm2.BitBtn1Click(Sender: TObject);
var
  klipperEstCom: string;
begin
  if form2.CheckBox1.Checked then
    klipperEstCom:=GetEnvironmentVariable('WINDIR')+'\System32\cmd.exe /C '+ExtractFileDir(Application.ExeName)+'\klipper_estimator.exe --config_moonraker_url '+edit1.Text+' --config_moonraker_api_key ' +edit2.Text+' dump-config > '+ExtractFileDir(Application.ExeName)+'\config.json'
    else klipperEstCom:=GetEnvironmentVariable('WINDIR')+'\System32\cmd.exe /C '+ExtractFileDir(Application.ExeName)+'\klipper_estimator.exe --config_moonraker_url '+edit1.Text+' dump-config > '+ExtractFileDir(Application.ExeName)+'\config.json';
  ExecNewProcess(klipperEstCom,true);
  if FileExists(ExtractFileDir(Application.ExeName)+'\config.json')=true then
    if FileSize(ExtractFileDir(Application.ExeName)+'\config.json')>0 then showmessage('Printer File Saved !')
      else showmessage('Error ! File not Saved !')
  else showmessage('Error ! File not Saved !');
end;
```
### Edited Interface
[image]

StimeKe, 2024-11-03:
Oh! Thank you so much for your prompt response. I have tested this version, and it's absolutely perfect. I am now able to connect to my Klipper.

closed 2024-11-03 by StimeKe