sb53systems/G-Code-Flow-Temperature-Controller #19: [Keeping Notes] getting the script working on Linux

Nathan22211, 2024-11-08:
at least until he figures out how to compile pascal for linux I've done some fiddling around with WineZGUI and got the script to launch thus far under a wine runner.

you'll need WineZGUI and possibly bottles, (both from flathub)

firstly you'll need import a recent version of the soda runner into WineZGUI, (or install it from that app). Bottles comes with soda preinstalled after running it once, so you can import it from there.

second, you'll want to right click on the Sb53 exe and have it run under wineZGUI. it'll ask you if you want to install some .NET core stuff, do that.

third, make a gamedir of the wine prefix folder through WineZGUI, otherwise you can't bring in the config folder nor the other exe.

lastly, have WineZGUI open the folder for the script's wine prefix via WineZGUI and drag the other needed stuff into it.

you should be able to run it standalone now but I haven't figured out how to run it from the slicer yet.

sb53systems, 2024-11-08:
@Nathan22211 ,Soon, I'll have time to compile the source code for Linux and macOS. It's not too difficult with Delphi 12, which allows you to compile the same code for any operating system. I recommend trying this, as it's the only way to make it work correctly! I'm here to help if needed.  
  
My discord is : sb53systems

Nathan22211, 2024-11-08:
I'll have to look to see where I could get Delphi 12 for arch linux, my main concern is that it might not behave correctly with the linux version of klipper estimator

anyway I'm one step closer. go into it's behavior under more and have it launch directly, you can then use `bash -c ''\''/home/nathan/.var/app/io.github.fastrizwaan.WineZGUI/data/winezgui/Prefixes/SB53-Systems-d8ebb469c5/klipper_estimator.sh'\'''`

replace `nathan` with your home user

Nathan22211, 2024-11-08:
ok that didn't work *sigh* well it's half done. I'm having trouble finding stuff on pascal compiling on arch though

sb53systems, 2024-11-08:
this could be done from any operating system (Windows for my case).
i just need to install a virtual Linux to be able to test it.  
  
I will send you a trial version if you can test it.

sb53systems, 2024-11-09:
After doing some research, I found that a paid version of Delphi 12 (Enterprise or Architect) is required to compile the code for Linux. With the Community edition, it is only possible to compile it for Windows, MacOS, or Android.  
  
I think another free IDE that runs on Linux can do this!  
[Lazarus IDE :](https://www.lazarus-ide.org/)
Lazarus is an open source integrated development environment (IDE) that allows programming in Pascal, particularly with the Free Pascal compiler. It is designed to create cross-platform applications and is compatible with Windows, macOS, and `Linux` operating systems.  
  
`Klipper Estimator` can also be compiled for `Linux`: [See the Klipper_Estimator Building Section](https://github.com/Annex-Engineering/klipper_estimator?tab=readme-ov-file#building) (This must be done for the modified version that is used by the Script [(This one)](https://github.com/sb53systems/klipper_estimator)).

sihawken, 2025-02-17:
@Nathan22211 

I got it working with regular wine + winetricks on debian.

winetricks -q win10
winetricks corefonts dotnet48

Honestly I don't even know if the dotnet48 is needed, but that is what worked for me.

Nathan22211, 2025-02-17:
I'll have to try it on arch when I'm home. I'm used to using wine via Bottles so not sure where I'd begin setting this up with regular wine

sihawken, 2025-02-17:
Here is how I would approach it on arch, partially based on this guide (https://wine.htmlvalidator.com/install-wine-on-arch-linux.html)

Enable multilib 

Make sure that multilib is enabled on arch:

`sudo nano /etc/pacman.conf`
```
[multilib]
Include = /etc/pacman.d/mirrorlist
```

Do a full system upgrade:

`sudo pacman -Syu`

Install wine (not sure if wine-mono or wine-gecko is necessary, but it can't hurt)

`sudo pacman -S wine wine-mono wine-gecko`

Download the latest winetricks:

```
wget  https://raw.githubusercontent.com/Winetricks/winetricks/master/src/winetricks
chmod +x winetricks 
sudo mv winetricks /usr/local/bin
```
Configure winetricks for windows 10, install corefonts and dotnet48

```
winetricks -q win10
winetricks corefonts dotnet48
```

Once that is done, you should be able to add it just like any regular post processing script, just with /bin/wine/ in front of it.

Nathan22211, 2025-03-26:
@sihawken didn't work for me, none of the buttons would respond and it's caused plasma shell to crash a few times