sb53systems/G-Code-Flow-Temperature-Controller #20: Problem Autogenerating PrinterConfig

Woolfie-maker, 2024-11-16:
Printer: QIDI X-MAX 3 (klipper based)

Issue: I load my printer config file from my printer's URL, and a message pops up saying file is saved however the contents of the config to JSON file do not update, and the date modified of the file also indicates that no changes actually saved.

I have manually looked through the config file of my printer and some terms listed within the json file are labelled differently within my printer config file. is this possibly the issue? I attempted to manually edit the terms and associated values within the json file which I probably did wrong however this didn't appear to help.

[printer.txt](https://github.com/user-attachments/files/17783947/printer.txt)

Please advise me on how to solve this issue.

sb53systems, 2024-11-16:
For some printers running a Klipper firmware limited by the manufacturer, the configuration cannot be retrieved via IP.  
You need to manually edit the provided "config.json" example file.   
  
Attached is the modified file based on your printer's configuration.  
[config.json](https://github.com/user-attachments/files/17784401/config.json)  
  
Let me know if you need more information.  
  
Best regards,  
Salim.

Woolfie-maker, 2024-11-16:
Thankyou, you got it working for me

sb53systems, 2024-11-16:
Congratulations, don't forget to give a star to the Github project.  
  
Happy Smart 3D Printing.

Woolfie-maker, 2024-11-16:
[image]
just one extra query.
Why is it that the program doesn't target 240celcius and my max set extrusion of 19mm^3/s in the area that I highlighted?
In my mind there should be no reason that it doesn't creep up to the 19mm^3 over a course of 20 seconds or so with my temperature capacity being set at 2 degrees per second

sb53systems, 2024-11-16:
This is normal because the script adjusts the temperature based on the average flow rate (the blue curve), even if some moves can reach the maximum flow. However, due to acceleration variations, the average would be lower.  
  
Let me know if you need further.

Woolfie-maker, 2024-11-16:
Ok, so its because the printer's start and end of each layer is at 0mm/s/ and that the printer has to spend time accelerating from this point?
So as layer times tend to infinity then the average flow rate will tend to the maximum rated flow at the maximum rated temperature?

sb53systems, 2024-11-17:
> Ok, so its because the printer's start and end of each layer is at 0mm/s/ and that the printer has to spend time accelerating from this point? So as layer times tend to infinity then the average flow rate will tend to the maximum rated flow at the maximum rated temperature?

Not exactly, this is caused during acceleration for each line.  
Be aware that the Script calculates the Average Flow for each second of printing and not for each layer.