sb53systems/G-Code-Flow-Temperature-Controller #21: Compile for Linux?

cray12399, 2024-11-20:
I found out about your awesome script from a Youtube video and want to give it a try. However, I use Linux and you compiled it as an exe. Would you be able to compile it as a Linux executable?

sb53systems, 2024-11-20:
Unfortunately not yet, the free version of the compiler used does not allow to generate executables for Linux, in addition to the fact that I am not an expert in programming!
Soon I will try to make a version under python that could be executed on all operating systems.  
  
Best regards,  
Salim.

cray12399, 2024-11-20:
Hi Salim,

What language are you writing the script in? If you provide the actual source code, I might be able to compile it for Linux myself.

sb53systems, 2024-11-20:
My actual source code:  
https://github.com/sb53systems/G-Code-Flow-Temperature-Controller/tree/main/Source/V1.1  
  
Compiled with Delphi 12 Community Edition (Free Pascal IDE)  
You will need a PRO or Entreprise Edition to compil for Linux.  
  
You can reach me out via Discord: sb53systems