sb53systems/G-Code-Flow-Temperature-Controller #22: An unavailable error occurred

RiceCat7487, 2024-12-29:
[image]
I followed the instructions step by step, but this error still occurred.
Is it because of my computer operating system?

sb53systems, 2024-12-29:
Hey,  
This error message indicates that a required file could not be found !   

- At the beginning send me the script folder Screenshot.  
[image]  
  
- Tell me more about your OS, the script uses the klipper_Estimator script, using ...\Windows\System32\CMD.exe. If not found this can make the same issue.  
  
Reach me out in discord for a better support. (desktop sharing will make it faster).  
  
Salim.

RiceCat7487, 2025-01-15:
[image]
[report.zip](https://github.com/user-attachments/files/18419642/report.zip)
Thank you for your reply. 
The system report and screenshots are included. I will try the solution you mentioned. If it still doesn’t work, I will contact you on Discord. 
Thank you again.

Lu.

sb53systems, 2025-01-15:
The path to the script must not have spaces, usually you have to place the folder containing the script directly in the local disk "D:\" or "C:\". Since you don't have a local disk "D:/", put it in your disk "C:\".  
Example "C:\SB53_G-Code_Flow_Temperature_Controller_V1.1".  
I think this will solve the problem. Anyway, keep me informed.  
  
Best regards  
Salim.

RiceCat7487, 2025-01-20:
[image]

Oh, this works. Thank you very much. I will solve the remaining problems slowly by myself. B
y the way, Happy Chinese New Year (New Year's Eve is on the 28th)

Best regards
Lu

closed 2025-01-20 by sb53systems

sb53systems, 2025-01-20:
I'm glad that you found the issue, Happy new year.  
Don't forget to give a Start to the project.  

Best regards  
Salim.