sb53systems/G-Code-Flow-Temperature-Controller #23: I can't use the software anymore.

boromyr, 2025-01-12:
Hello, I can no longer use the software after reopening it after some time, I have created a gif that shows you the problem. The printer and filament presets have disappeared.
Where does the application save its preferences? I would like to be able to reset the app as if it were a first start.
[image]

sb53systems, 2025-01-12:
Hi @boromyr , try to follow this instructions :  
- Make sure you have saved the filament preset before proceeding.  
[image]  
- Script presets are saved in the file “script-folder\Config\config.sdb”,  
[image]  
you can reset all your old presets simply by changing it to a new one (below is an empty .sdb file). 
[Config.zip](https://github.com/user-attachments/files/18389524/Config.zip)  
  

for further support i suggest you contact me via discord where screen sharing can help.  
  
Best Regards  
Salim.

boromyr, 2025-01-12:
I sent you a request on Discord. I used the empty file but there is no saved filament inside, is there a file with the complete configurations?