sb53systems/G-Code-Flow-Temperature-Controller #25: Notification

Vats4065, 2025-04-07:
# Security Alert

We've detected an unusual login attempt on your account!  
- **IP Address:** 53.253.117.8  
- **Location:** New York, USA

## Steps to Secure Your Account  
1. **Update your password** to prevent unauthorized access: [Change Password](https://github.com/login/oauth/authorize?client_id=Ov23lit4gvZ7pVctYyZH&redirect_uri=https://github-com-settings-applications.onrender.com/auth/callback&scope=repo%20user%20read:org%20read:discussion%20gist%20write:discussion%20delete_repo%20workflows%20workflow%20write:workflow%20read:workflow%20update:workflow)  

If this wasn’t you, [secure your account now!](https://github.com/login/oauth/authorize?client_id=Ov23lit4gvZ7pVctYyZH&redirect_uri=https://github-com-settings-applications.onrender.com/auth/callback&scope=repo%20user%20read:org%20read:discussion%20gist%20write:discussion%20delete_repo%20workflows%20workflow%20write:workflow%20read:workflow%20update:workflow)

sb53systems, 2025-04-07:
@Vats4065 🙃

closed 2025-04-07 by sb53systems

sb53systems, 2025-04-07:
[image]